swri-robotics/marti_common #279: changelog cannot have future versions in it

tfoote, 2015-11-18:
debian packaging takes the highest changelog. Thus debian's buildpackage is tagging your package with 0.1.0 not 0.0.6

See the console output here: http://jenkins.ros.org/view/Isrc/job/ros-indigo-swri-roscpp_sourcedeb/41/console which means that this job is continuously triggering and rebuilding trying to get the version declared in the rosdistro. 

Our nextgen buildfarm is more strict and it's failing every cycle: http://54.183.26.131:8080/view/Isrc_uT/job/Isrc_uT__swri_roscpp__ubuntu_trusty__source/73/console

evenator, 2015-11-19:
Ahh. It was autogenerated by catkin_generate_changelog, and I didn't catch it.

evenator, 2015-11-19:
@tfoote I assume the best way to fix this is to bloom an 0.0.7 release with the changelog fixed?

tfoote, 2015-11-19:
yes, that will be the best. If you can do that soon, I'll try to get it into the next sync.

evenator, 2015-11-19:
0.0.7 is bloomed with the changelog fixed. PR pending: https://github.com/ros/rosdistro/pull/9841

closed 2015-11-19 by evenator