swri-robotics/marti_common #280: initialize_origin.py not installed

Georacer, 2015-11-25:
Hello,

I try to invoke the initialize_origin.py script from my launch file as such:
```xml
    <node pkg="swri_transform_util" type="initialize_origin.py" name="initialize_origin" >
        <param name="local_xy_frame" value="/map"/>
        <param name="local_xy_origin" value="Spata"/>
        <rosparam param="local_xy_origins">
            [{ name: Spata,
             latitude: 37.9879772,
             longitude: 23.9078602,
             altitude: 129.69,
             heading: 0.0}]
        </rosparam>
    </node>
```

I get the error:
```bash
ERROR: cannot launch node of type [swri_transform_util/initialize_origin.py]: can't locate node [initialize_origin.py] in package [swri_transform_util]
```
catkin can't see the python script as executable.

I fixed it by `chmod +x` the .py script, but it's not ideal.
I noticed that you recently added an install target to the script but perhaps it doesn't work?

I have the latest version of `marti_common` git pulled in my workspace.

pjreed, 2015-11-25:
Hmm... when I build and install swri_transform_utils in my own workspace, it looks like initialize_origin.py is installed and marked as executable in install/lib/swri_transform_util/initialize_origin.py .  Also, after installing the official ros-indigo-swri-transform-util package, it is installed and executable at /opt/ros/indigo/lib/swri_transform_util/initialize_origin.py

In your clone of the repository, is the file present & exectuable in marti_common/swri_transform_util/nodes/ ?  Also, does the CMakeLists.txt file for swri_transform_util have a catkin_install_python call for initialize_origin.py?

Georacer, 2015-11-25:
Ah, I have a guess on what's wrong.

I needed to format my laptop yesterday. I backed up and copy/pasted the `marti_common` folder in my new workspace.

However, the executable was assigned to my previous user account and hence the .py file was not executable any more.

closed 2015-11-25 by Georacer