swri-robotics/marti_common #303: Travis CI Build Fails Erroneously for geometry_util on jade-devel

evenator, 2016-03-05:
Compare these CI builds of the same commit, and you can see that [our Travis CI fails](https://travis-ci.org/swri-robotics/marti_common/builds/113832798) and [the OSRF Jenkins build passes](http://build.ros.org/job/Jdev__marti_common__ubuntu_trusty_amd64/8/).

This package also seems to build fine for me on Mint 17.2.

130s, 2016-03-07:
[I did an experiment to use format 1 for the package that's causing an error](https://github.com/130s/marti_common/pull/1) and Travis passed for my forked repository. So it may be how `catkin_tools` handles package format 2.

evenator, 2016-03-08:
Interesting. I'll have to look into that.

pjreed, 2016-04-11:
So there were actually two issues here, and I fixed them both in #311.

One is a failure to find Eigen3, and I spent a while beating my head against that without figuring why it was happening.  It always worked in my Jade VMs but failed in the Travis build.  I "fixed" it by adding a fallback that will use PkgConfig to find Eigen3 if the CMake Find fails.  That seems to work, so it's good enough for me.

The second issue was the failing geometry_util test, which was because swri_image_util was erroneously installing the geometry_util.test file in its "launch" directory; the Travis build was them trying to run the test out of the install dir, but it would fail because the compiled cpp code for the test did not get installed.  I fixed it by not installing that .test file.

closed 2016-04-11 by pjreed

130s, 2016-05-01:
@pjreed 

> So there were actually two issues here, and I fixed them both in #311.

You mean https://github.com/swri-robotics/marti_common/pull/309?

pjreed, 2016-05-02:
Oops, that is correct.