swri-robotics/marti_common #348: FindGeos.cmake fails on Ubuntu Xenial

evenator, 2016-05-24:
Because the path to libgeos is now `/usr/lib/x86_64-linux-gnu/geos-3.5.0`, the regex to find the library name matches on a portion of the library directory, resulting in broken linker flags:

```
[swri_geometry_util:cmake] -- Found GEOS: inux-gnugeos-3.5.0 (version: 3.5.0, includes: /usr/include)
```
...
```
[swri_geometry_util:make] [100%] Linking CXX shared library /root/ros/ws_ci_src/devel/.private/swri_geometry_util/lib/libswri_geometry_util.so
[swri_geometry_util:make] /usr/bin/cmake -E cmake_link_script CMakeFiles/swri_geometry_util.dir/link.txt --verbose=1
[swri_geometry_util:make] /usr/bin/c++  -fPIC   -shared -Wl,-soname,libswri_geometry_util.so -o /root/ros/ws_ci_src/devel/.private/swri_geometry_util/lib/libswri_geometry_util.so CMakeFiles/swri_geometry_util.dir/src/intersection.cpp.o CMakeFiles/swri_geometry_util.dir/src/polygon.cpp.o CMakeFiles/swri_geometry_util.dir/src/geometry_util.cpp.o CMakeFiles/swri_geometry_util.dir/src/cubic_spline.cpp.o /opt/ros/kinetic/lib/libtf.so /opt/ros/kinetic/lib/libtf2_ros.so /opt/ros/kinetic/lib/libactionlib.so /opt/ros/kinetic/lib/libmessage_filters.so /opt/ros/kinetic/lib/libroscpp.so -lboost_signals -lboost_filesystem /opt/ros/kinetic/lib/libxmlrpcpp.so /opt/ros/kinetic/lib/libtf2.so /opt/ros/kinetic/lib/libroscpp_serialization.so /opt/ros/kinetic/lib/librosconsole.so /opt/ros/kinetic/lib/librosconsole_log4cxx.so /opt/ros/kinetic/lib/librosconsole_backend_interface.so -llog4cxx -lboost_regex -lboost_chrono -lboost_atomic /opt/ros/kinetic/lib/librostime.so -lboost_date_time /opt/ros/kinetic/lib/libcpp_common.so -lboost_system -lboost_thread -lpthread -lconsole_bridge /usr/lib/x86_64-linux-gnu/libopencv_videostab.so.2.4.9 /usr/lib/x86_64-linux-gnu/libopencv_superres.so.2.4.9 /usr/lib/x86_64-linux-gnu/libopencv_stitching.so.2.4.9 /usr/lib/x86_64-linux-gnu/libopencv_ocl.so.2.4.9 /usr/lib/x86_64-linux-gnu/libopencv_gpu.so.2.4.9 /usr/lib/x86_64-linux-gnu/libopencv_contrib.so.2.4.9 -linux-gnugeos-3.5.0 /usr/lib/x86_64-linux-gnu/libopencv_photo.so.2.4.9 /usr/lib/x86_64-linux-gnu/libopencv_legacy.so.2.4.9 /usr/lib/x86_64-linux-gnu/libopencv_video.so.2.4.9 /usr/lib/x86_64-linux-gnu/libopencv_objdetect.so.2.4.9 /usr/lib/x86_64-linux-gnu/libopencv_ml.so.2.4.9 /usr/lib/x86_64-linux-gnu/libopencv_calib3d.so.2.4.9 /usr/lib/x86_64-linux-gnu/libopencv_features2d.so.2.4.9 /usr/lib/x86_64-linux-gnu/libopencv_highgui.so.2.4.9 /usr/lib/x86_64-linux-gnu/libopencv_imgproc.so.2.4.9 /usr/lib/x86_64-linux-gnu/libopencv_flann.so.2.4.9 /usr/lib/x86_64-linux-gnu/libopencv_core.so.2.4.9 -Wl,-rpath,/opt/ros/kinetic/lib: 
[swri_geometry_util:make] /usr/bin/ld: cannot find -linux-gnugeos-3.5.0
```

davidrotor, 2016-07-05:
I have this problem how do I fix?

evenator, 2016-07-05:
This problem was fixed in kinetic-devel. See 9923bc8d94721987ef28064e6be8690bf60bc744

closed 2016-07-05 by evenator