swri-robotics/marti_common #411: swri_yaml_util Fails Current Version of Industrial CI

evenator, 2017-02-17:
I recently discovered that swri_yaml_util is failing CI where it previously was passing (see #410). Since swri_yaml_util hadn't changed, I deduced that Industrial CI had changed their CI scripts. It turns out that when ros-industrial/industrial_ci#116 started running all tests in docker, they set the source space in the docker image to read-only. This is a problem for swri_yaml_util, because it creates a file (version.h) in the source directory.

I'm inclined to say that Industrial CI's behavior (forcing read-only on the source directory) is correct here. Catkin builds are supposed to be out-of-source. We should fix swri_yaml_util.

evenator, 2017-02-17:
@130s FYI

130s, 2017-02-17:
Yes, another user is affected and we're discussing at https://github.com/ros-industrial/industrial_ci/issues/122 and https://github.com/ros-industrial/industrial_ci/pull/123
CC @ipa-mdl

ipa-mdl, 2017-02-17:
According to CMake conventions these generated files should be kept in `${PROJECT_BINARY_DIR}` or `${CMAKE_CURRENT_BINARY_DIR}` (this is somewhere within the build space)
For catkin-packages devel space is fine as well.
This way you can build the same src linked into different workspace without any collisions.

PR #412 (2017-03-02): Make swri_yaml_util build out-of-source
Fixes #411 by generating version.h in the devel space include folder instead of the source space.
Based heavily on http://answers.ros.org/question/123221/
commit e9bdbf6a (2017-03-02): Make swri_yaml_util build out-of-source

Fixes #411 by generating version.h in the devel space include folder instead of the source space.
Based heavily on http://answers.ros.org/question/123221/
closed 2017-03-02 by malban