swri-robotics/marti_common #444: swri_transform_util handles "/wgs84" but not "wgs84"

pjreed, 2017-05-12:
See swri-robotics/mapviz#478.  I would expect `swri_transform_util` to handle whatever its `wgs84_frame` is set to regardless of whether it has a slash in front of it or not.

danielsnider, 2017-05-13:
This issue prompts the question, does it make sense to allow "utm" in addition to "/utm" as well? How about "/local_xy" too?

Relevent file: https://github.com/swri-robotics/marti_common/blob/indigo-devel/swri_transform_util/include/swri_transform_util/frames.h

evenator, 2017-05-13:
👍 For adding support to the non-slash, but I don't think we should change the constants--that just creates confusion on the user end.

I recommend implementing a custom comparison function in `swri_transform_util` that does TF frame comparison and matches regardless of the leading slash. That way we can guarantee consistent behavior across the whole codebase, and I can think of other places I've had to compare TF frames that it would be useful.

evenator, 2017-05-13:
`tf` uses `tf::strip_leading_slash()` before all comparisons for this reason. That might be the answer here as well. I still prefer the idea of implementing a comparison function, but that might just be because I wrote too much Java in my youth.

pjreed, 2017-05-15:
While implementing something like `String.compareIgnoreLeadingSlash()` would make me happy, for the sake of consistency it's probably better to do it just like `tf` does it.

PR #492 (2017-12-06): Normalize TF frames before comparisons.
- Add function to normalize leading slash in TF frames
- Add function to compare normalized TF frames
- Use normalized comparisons for all TF frame comparisons

Closes #444
closed 2017-12-06 by pjreed