swri-robotics/marti_common #453: Finish Branch Unification

evenator, 2017-06-19:
- [ ] Sync master branch with indigo-devel
- [ ] Set master as the default branch
- [ ] Update readme
- [ ] Delete `jade-devel` (replace with tag)
- [ ] Migrate SwRI's internal projects to the master branch (or to Debian installations)
- [ ] Delete `kinetic-devel` (replace with tag)
- [ ] Delete `indigo-devel` (replace with tag)

pjreed, 2017-06-20:
MRs have been submitted in GitLab for the affected workspace files.

pjreed, 2017-06-20:
I've submitted PRs to release `0.3.0` of `marti_common` into indigo, jade, kinetic, and lunar:

- ros/rosdistro#15312
- ros/rosdistro#15313
- ros/rosdistro#15315
- ros/rosdistro#15316

pjreed, 2017-06-20:
PRs have been accepted, so I'll close this.

closed 2017-06-20 by pjreed