swri-robotics/marti_common #468: [swri_transform_util] Flaky Tests

evenator, 2017-07-28:
As you can see in [this CI build](https://travis-ci.org/swri-robotics/marti_common/jobs/258669229), there are some tests in `swri_transform_util` that seem to fail at random.

evenator, 2017-07-28:
I retriggered the linked build, so it will probably appear passed. However, someone should look into what causes a test failure like this:

```
[ROSTEST]-----------------------------------------------------------------------

[swri_transform_util.rosunit-test_transform_manager/Identity1][passed]

[swri_transform_util.rosunit-test_transform_manager/Identity2][passed]

[swri_transform_util.rosunit-test_transform_manager/TfToTf1][passed]

[swri_transform_util.rosunit-test_transform_manager/TfToTf2][passed]

[swri_transform_util.rosunit-test_transform_manager/WgsToUtm][FAILURE]----------

/root/catkin_ws/src/marti_common/swri_transform_util/test/test_transform_manager.cpp:130

Value of: _tf_manager.GetTransform( swri_transform_util::_utm_frame, swri_transform_util::_wgs84_frame, transform)

  Actual: false

Expected: true

--------------------------------------------------------------------------------

[swri_transform_util.rosunit-test_transform_manager/UtmToWgs84][FAILURE]--------

/root/catkin_ws/src/marti_common/swri_transform_util/test/test_transform_manager.cpp:152

Value of: _tf_manager.GetTransform( swri_transform_util::_wgs84_frame, swri_transform_util::_utm_frame, transform)

  Actual: false

Expected: true

--------------------------------------------------------------------------------

[swri_transform_util.rosunit-test_transform_manager/TfToUtm1][FAILURE]----------

/root/catkin_ws/src/marti_common/swri_transform_util/test/test_transform_manager.cpp:174

Value of: _tf_manager.GetTransform( swri_transform_util::_utm_frame, "/far_field", transform)

  Actual: false

Expected: true

--------------------------------------------------------------------------------

[swri_transform_util.rosunit-test_transform_manager/TfToUtm2][FAILURE]----------

/root/catkin_ws/src/marti_common/swri_transform_util/test/test_transform_manager.cpp:195

Value of: _tf_manager.GetTransform( swri_transform_util::_utm_frame, "/far_field", transform)

  Actual: false

Expected: true

--------------------------------------------------------------------------------

[swri_transform_util.rosunit-test_transform_manager/UtmToTf1][FAILURE]----------

/root/catkin_ws/src/marti_common/swri_transform_util/test/test_transform_manager.cpp:217

Value of: _tf_manager.GetTransform( "/far_field", swri_transform_util::_utm_frame, transform)

  Actual: false

Expected: true

--------------------------------------------------------------------------------

[swri_transform_util.rosunit-test_transform_manager/UtmToTf2][FAILURE]----------

/root/catkin_ws/src/marti_common/swri_transform_util/test/test_transform_manager.cpp:239

Value of: _tf_manager.GetTransform( "/near_field", swri_transform_util::_utm_frame, transform)

  Actual: false

Expected: true

--------------------------------------------------------------------------------

[swri_transform_util.rosunit-test_transform_manager/UtmToTf3][FAILURE]----------

/root/catkin_ws/src/marti_common/swri_transform_util/test/test_transform_manager.cpp:261

Value of: _tf_manager.GetTransform( "/far_field", swri_transform_util::_utm_frame, transform)

  Actual: false

Expected: true

--------------------------------------------------------------------------------

[swri_transform_util.rosunit-test_transform_manager/UtmToTf4][FAILURE]----------

/root/catkin_ws/src/marti_common/swri_transform_util/test/test_transform_manager.cpp:283

Value of: _tf_manager.GetTransform( "/far_field", swri_transform_util::_utm_frame, transform)

  Actual: false

Expected: true

--------------------------------------------------------------------------------

[swri_transform_util.rosunit-test_transform_manager/Wgs84ToTf1][FAILURE]--------

/root/catkin_ws/src/marti_common/swri_transform_util/test/test_transform_manager.cpp:305

Value of: _tf_manager.GetTransform( "/far_field", swri_transform_util::_wgs84_frame, transform)

  Actual: false

Expected: true

--------------------------------------------------------------------------------

[swri_transform_util.rosunit-test_transform_manager/Wgs84ToTf2][FAILURE]--------

/root/catkin_ws/src/marti_common/swri_transform_util/test/test_transform_manager.cpp:327

Value of: _tf_manager.GetTransform( "/near_field", swri_transform_util::_wgs84_frame, transform)

  Actual: false

Expected: true

--------------------------------------------------------------------------------

[swri_transform_util.rosunit-test_transform_manager/TfToWgs84_1][FAILURE]-------

/root/catkin_ws/src/marti_common/swri_transform_util/test/test_transform_manager.cpp:349

Value of: _tf_manager.GetTransform( swri_transform_util::_wgs84_frame, "/far_field", transform)

  Actual: false

Expected: true

--------------------------------------------------------------------------------

[swri_transform_util.rosunit-test_transform_manager/TfToWgs84_2][FAILURE]-------

/root/catkin_ws/src/marti_common/swri_transform_util/test/test_transform_manager.cpp:370

Value of: _tf_manager.GetTransform( swri_transform_util::_wgs84_frame, "/near_field", transform)

  Actual: false

Expected: true

--------------------------------------------------------------------------------

SUMMARY

 * RESULT: FAIL

 * TESTS: 16

 * ERRORS: 0

 * FAILURES: 12
```

evenator, 2017-07-30:
I think the problem may be that the tests are sometimes running before the origin is received from `initialize_origin.py`.

malban, 2017-08-02:
Fixed in #469

closed 2017-08-02 by malban

evenator, 2017-08-02:
Sidenote: it looks like this is a duplicate of #304