swri-robotics/marti_common #482: Fix Catkin Lint

evenator, 2017-08-29:
There are numerous instances of Catkin errors and warnings in this repository. There's now a CI check for them that's allowed to fail. Fix the errors and make the CI check required.

Note: It looks like `catkin_lint` can't expand macros, so it throws spurious errors about installing non-existent targets for `swri_nodelet_add_node` targets.

pjreed, 2017-08-29:
Also note that calls to `catkin_install_python` for installing Python scripts are flagged for the same reason.

evenator, 2017-08-30:
> Also note that calls to catkin_install_python for installing Python scripts are flagged for the same reason.

That's especially odd, since `catkin_install_python` is the preferred way of installing Python scripts. I suspect this is fixable, unlike the general case, which would require adding support for `include()` and macro expansion.