swri-robotics/marti_common #495: [swri_transform_manager]: problems when use_sim_time=true

facontidavide, 2017-11-03:
Hi,

I detected this problem using Mapviz, but I guess it should be reported here. if it is not the case, let me know.

I am trying to use the time_map plugin of Mapviz playing a rosbag file.
If I use use_sim_time=true, the transfom from __/far_field__ to __/wgs84__ fails here

https://github.com/swri-robotics/marti_common/blob/master/swri_transform_util/src/transform_manager.cpp#L211

If there any way to fix this?

facontidavide, 2017-11-03:
I am closing this issue. I MIGHT have found the problem and it is not related to your library...

closed 2017-11-03 by facontidavide

evenator, 2017-11-03:
Can you please post the results of `rosrun tf view_frames`?

facontidavide, 2017-11-03:
The problem was that I forgot to play the rosbag with --clock.

Furthermore, I had many problems with another bag, probably related to this bug:
https://github.com/introlab/rtabmap_ros/issues/170

I was using a parsed bag and the tf are messed up.

As I said, eventually this is not a problem related to swri_transform_manager nor mapviz.