swri-robotics/marti_common #502: swri_route_util: lateral_acceleration_mode Parameter Must Be 0 or 1

evenator, 2017-11-20:
Because `boost::lexical_cast<int>()` is used to parse the value of `lateral_acceleration_mode`, it must be "0" or "1". (See https://github.com/swri-robotics/marti_common/blob/40fc462700ce32c52062e698ce526174539d1b61/swri_route_util/src/route_speeds.cpp#L99)

This would also be true if `boost::lexical_cast<bool>()` were used. The following values should be acceptable:

- "true"
- "True"
- "1" (currently works)
- "false"
- "False"
- "0" (currently works)

Note that all of these values work with ROS parameters.

evenator, 2017-11-20:
I did some testing with Boost lexical cast and Standard Library string streams, (see https://gist.github.com/evenator/86e6404e1f9e446c062ef650b496be4c). Here are the results:

```
Error during boost lexical cast of true
Error during boost lexical cast of True
Error during boost lexical cast of TRUE
Boost lexical cast of 1 = true
Boost lexical cast of 1 = true
Error during boost lexical cast of false
Error during boost lexical cast of False
Error during boost lexical cast of FALSE
Boost lexical cast of 0 = false
Boost lexical cast of 0 = false
String stream cast of true = false
String stream cast of True = false
String stream cast of TRUE = false
String stream cast of 1 = true
String stream cast of false = false
String stream cast of False = false
String stream cast of FALSE = false
String stream cast of 0 = false
```

In order to use true/false strings, we'll need to parse them explicitly.

evenator, 2017-11-20:
It would be a good idea to implement this in `swri_string_util`, which already has functions for extracting various numerical types from strings.

malban, 2017-11-21:
I think this is a good idea.  I had a similar issue in route_point.cpp a while back with the "stop_point" field.  I like targeting consistency with ROS parameters.

Well, swri_string_util is kind of ridiculous in that it is a single file with 4 string utility functions, but it already exists, so I guess we may as well use it.  Maybe one day it will grow up to be a real package.