swri-robotics/marti_common #508: Create tool to debug tf frames coming from multiple publishers

pjreed, 2017-12-18:
An issue we've run into several times is when multiple different nodes are publishing a tf frame.  This often manifests as a frame shifting or rotating in periodic intervals in visualization tools.  While there's nothing illegal about a frame coming from multiple different sources, in practice this is very rare and almost never a useful case.

It'd be nice to have a node or a script that could run and alert you when different sources have published a frame and also identify what those sources are.

@matt-attack @kriskozak

evenator, 2017-12-19:
I believe the [roswtf](http://wiki.ros.org/tf#roswtf_plugin) plugin does this, but it's a tool, not a monitoring node.

matt-attack, 2017-12-20:
I messed around with roswtf on my system and it appears that it does indicate if there are multiple publishers

pjreed, 2017-12-20:
Oh, that's nice.  Does it indicate who the publishers are?  I've always had an issue with `tf_monitor` that it never seemed to be able to identify who was publishing what...

matt-attack, 2017-12-20:
It appears so.
```Beginning tests of your ROS graph. These may take awhile...
analyzing graph...
... done analyzing graph
running graph rules...
... done running graph rules
running tf checks, this will take a second...
... tf checks complete

Online checks summary:

Found 1 error(s).

ERROR TF multiple authority contention:
 * node [/static_tf2] publishing transform [/map] with parent [/far_field] already published by node [/static_tf]
 * node [/static_tf] publishing transform [/map] with parent [/far_field] already published by node [/static_tf2]
```