swri-robotics/marti_common #510: Error compiling. Bad copy and paste in CMakeLists.txt?

facontidavide, 2018-01-29:
Hi, 

The current master branch won't compile because in __swri_image_utils__ there are the following lines of code...

>   find_package(rostest REQUIRED)
>   add_rostest_gtest(test_image_util test/geometry_util.test test/test_geometry_util.cpp)
>   target_link_libraries(test_image_util ${PROJECT_NAME})

Cheers

G-Pereira, 2020-04-02:
What is the solution to this?

pjreed, 2020-04-02:
As far as I can tell, the current master branch compiles fine.  For reference:
http://build.ros.org/job/Mdev__marti_common__ubuntu_bionic_amd64/
https://travis-ci.org/github/swri-robotics/marti_common

If you're having an error, could you copy & paste the output of your build log?

G-Pereira, 2020-04-02:
CMake Error at /home/mendel/ros-melodic/install_isolated/share/rostest/cmake/rostest-extras.cmake:75 (add_executable):
  add_executable cannot create target "test_geometry_util" because another
  target with the same name already exists.  The existing target is an
  executable created in source directory
  "/home/mendel/ws/src/marti_common/swri_geometry_util".  See documentation
  for policy CMP0002 for more details.
Call Stack (most recent call first):
  /home/mendel/ros-melodic/install_isolated/share/rostest/cmake/rostest-extras.cmake:100 (_add_rostest_google_test)
  marti_common/swri_image_util/CMakeLists.txt:146 (add_rostest_gtest)

CMake Error at /home/mendel/ros-melodic/install_isolated/share/rostest/cmake/rostest-extras.cmake:76 (target_link_libraries):
  Attempt to add link library "gtest" to target "test_geometry_util" which is
  not built in this directory.
Call Stack (most recent call first):
  /home/mendel/ros-melodic/install_isolated/share/rostest/cmake/rostest-extras.cmake:100 (_add_rostest_google_test)
  marti_common/swri_image_util/CMakeLists.txt:146 (add_rostest_gtest)

CMake Error at marti_common/swri_image_util/CMakeLists.txt:147 (target_link_libraries):
  Attempt to add link library "swri_image_util" to target
  "test_geometry_util" which is not built in this directory.

G-Pereira, 2020-04-02:
I am using ROS Melodic Morenia on Debian 9

G-Pereira, 2020-04-02:
According to ROS I can also use the master branch for ROS Melodic. Is that still true?
https://index.ros.org/p/swri_image_util/github-swri-robotics-marti_common/#melodic

G-Pereira, 2020-04-02:
BTW I have libgtest-dev installed

pjreed, 2020-04-02:
The master branch does build on Melodic, and it looks like it's building on Debian on the build farm: http://build.ros.org/job/Mbin_ds_dS64__swri_image_util__debian_stretch_amd64__binary/

I think this is failing for you because `swri_image_util` builds a unit test that has the same target name as one of `swri_geometry_util`'s tests, but although I'm not sure why it's succeeding everywhere else.  I made a branch that renames it, so try this and see if it builds for you: https://github.com/pjreed/marti_common/tree/rename-image-util-test

G-Pereira, 2020-04-03:
Ok, thank you

PR #575 (2020-05-13): Rename swri_image_util test
The target name of this test was conflicting with a test in swri_geometry_util,
and that was causing builds to fail on some systems.

Fixes #510
commit 915a8952 (2020-05-13): Rename swri_image_util test (#575)

The target name of this test was conflicting with a test in swri_geometry_util,
and that was causing builds to fail on some systems.

Fixes #510
closed 2020-05-13 by pjreed