swri-robotics/marti_common #528: Add Python support for swri_transform_util

agyoungs, 2018-10-31:
Specifically at a minimum wgs84 transformations would be useful.  Here is the code used in the past:

```
def wgs84_to_local_xy(lat, lon, local_origin):
    earth_eccentricity = 0.08181919084261
    earth_equator_radius = 6378137.0
    reference_heading = 0.0  # in radians
    reference_latitude = local_origin.pose.position.y * math.pi / 180
    reference_longitude = local_origin.pose.position.x * math.pi / 180
    cos_heading = math.cos(reference_heading)
    sin_heading = math.sin(reference_heading)

    depth = -local_origin.pose.position.z

    p = earth_eccentricity * math.sin(reference_latitude)
    p = 1.0 - p * p

    rho_e = earth_equator_radius * (1.0 - earth_eccentricity * earth_eccentricity) / (math.sqrt(p) * p)
    rho_n = earth_equator_radius / math.sqrt(p)

    rho_lat = rho_e - depth
    rho_lon = (rho_n - depth) * math.cos(reference_latitude)

    rlat = lat * math.pi / 180
    rlon = lon * math.pi / 180
    dlat = (rlat - reference_latitude) * rho_lat
    dlon = (rlon - reference_longitude) * rho_lon

    y = dlat * cos_heading + dlon * sin_heading
    x = -1.0 * (dlat * sin_heading - dlon * cos_heading)

    return x, y

def local_xy_to_wgs84(x, y, local_origin):
    earth_eccentricity = 0.08181919084261
    earth_equator_radius = 6378137.0
    reference_heading = 0.0
    reference_latitude = local_origin.pose.position.y * math.pi / 180.0
    reference_longitude = local_origin.pose.position.x * math.pi / 180.0
    cos_heading = math.cos(reference_heading)
    sin_heading = math.sin(reference_heading)

    depth = -local_origin.pose.position.y

    p = earth_eccentricity * math.sin(reference_latitude)
    p = 1.0 - p * p

    rho_e_num = earth_equator_radius * (1.0 - earth_eccentricity * earth_eccentricity)
    rho_e_den = math.sqrt(p) * p
    rho_e = rho_e_num / rho_e_den
    rho_n = earth_equator_radius / math.sqrt(p)

    rho_lat = rho_e - depth
    rho_lon = (rho_n - depth) * math.cos(reference_latitude)

    dlon = cos_heading * x + sin_heading * y
    dlat = (y - dlon * sin_heading) / cos_heading
    rlat = dlat / rho_lat + reference_latitude
    rlon = dlon / rho_lon + reference_longitude

    lat = rlat * 180.0 / math.pi
    lon = rlon * 180.0 / math.pi
    return lat, lon
```

agyoungs, 2018-10-31:
The above snippet has been observed to be slow when iterating over a bunch of transforms.  The following code makes use of numpy arrays:

```
def wgs84_to_local_xy(lat, lon, local_origin):
        
        lat = np.array(lat)
        lon = np.array(lon)
        earth_eccentricity = 0.08181919084261
        earth_equator_radius = 6378137.0
        reference_heading = 0.0
        reference_latitude = local_origin['latitude'][0] * math.pi / 180.0
        reference_longitude = local_origin['longitude'][0] * math.pi / 180.0
        cos_heading = math.cos(reference_heading)
        sin_heading = math.sin(reference_heading)

        depth = -local_origin['altitude'][0]

        p = earth_eccentricity * math.sin(reference_latitude)
        p = 1.0 - np.multiply(p, p)

        rho_e_num = earth_equator_radius * (1.0 - earth_eccentricity * earth_eccentricity)
        rho_e_den = np.multiply(np.sqrt(p),  p)
        rho_e = np.divide(rho_e_num, rho_e_den)
        rho_n = np.divide(earth_equator_radius, np.sqrt(p))

        rho_lat = rho_e - depth;
        rho_lon = (rho_n - depth) * np.cos(reference_latitude);

        rlat = lat * math.pi / 180;
        rlon = lon * math.pi / 180;
        dlat = (rlat - reference_latitude) * rho_lat;
        dlon = (rlon - reference_longitude) * rho_lon;

        y = dlat * cos_heading + dlon * sin_heading;
        x = -1.0 * (dlat * sin_heading - dlon * cos_heading);

        return x, y

    def local_xy_to_wgs84(x, y, local_origin):

        x = np.array(x)
        y = np.array(y)
        earth_eccentricity = 0.08181919084261
        earth_equator_radius = 6378137.0
        reference_heading = 0.0
        reference_latitude = local_origin['latitude'][0] * math.pi / 180.0
        reference_longitude = local_origin['longitude'][0] * math.pi / 180.0
        cos_heading = math.cos(reference_heading)
        sin_heading = math.sin(reference_heading)

        depth = -local_origin['altitude'][0]

        p = earth_eccentricity * math.sin(reference_latitude)
        p = 1.0 - np.multiply(p, p)

        rho_e_num = earth_equator_radius * (1.0 - earth_eccentricity * earth_eccentricity)
        rho_e_den = np.multiply(np.sqrt(p), p)
        rho_e = np.divide(rho_e_num, rho_e_den)
        rho_n = np.divide(earth_equator_radius, np.sqrt(p));

        rho_lat = rho_e - depth;
        rho_lon = (rho_n - depth) * math.cos(reference_latitude);

        dLon = cos_heading * x + sin_heading * y;
        dLat = (y - dLon * sin_heading) / cos_heading;
        rlat = np.divide(dLat, rho_lat) + reference_latitude;
        rlon = np.divide(dLon, rho_lon) + reference_longitude;

        lat = rlat * 180.0 / math.pi;
        lon = rlon * 180.0 / math.pi;
        return lat, lon
```

evenator, 2018-12-19:
I have a WIP branch from ages ago that exposed the C++ library as a Python CModule. https://github.com/evenator/marti_common/tree/swri_transform_util-python-modules

agyoungs, 2019-05-23:
> I have a WIP branch from ages ago that exposed the C++ library as a Python CModule. https://github.com/evenator/marti_common/tree/swri_transform_util-python-modules

Good to know. If I ever get time, I'll take a look at it and try to finish it up

agyoungs, 2020-07-14:
#595 adds the simple transformer and fixes an issue of ignoring orientation of the local_xy origin. I still think it's useful to expose the c++ library as a Python CModule, however.

danthony06, 2023-09-29:
I'm going to close this. I think it's a neat idea, but I don't have the maintenance bandwidth to support this unless there's more demand for it. I'm happy to revisit it in the future.

closed 2023-09-29 by danthony06