swri-robotics/marti_common #601: Topic Services not working on ROS2 (foxy)

dvmartin999, 2020-09-12:
The swri-roscpp support for topic services under ROS2 (at least for foxy, but I'm pretty sure dashing as well) isn't working because of a change in message name generation.  The current python splitter expects <Service>Request.h and <Service>Response.h, but with ROS2, the IDL generator forces names into snake_case. The splitter needs to be updated accordingly.

pjreed, 2020-09-14:
@matt-attack, do you have time to look at this?  If not, this might be a good issue for @danthony06.

danthony06, 2020-09-14:
I can look at it.

dvmartin999, 2020-09-14:
@danthony06 I've got a fix I'm working with now, should be a couple days before I can verify.  I'm happy to provide as-is, or push when it's ready.  Just let me know how you want to proceed.

danthony06, 2020-09-14:
If you have a fix, I can work on verifying it.

dvmartin999, 2020-09-14:
Can you add me as a contributor, then I can just push my branch.  Or would you prefer I send by email?

danthony06, 2020-09-14:
Do you have the repo forked somewhere that I could branch off of?

dvmartin999, 2020-09-14:
Ok - forked, pushed my changes to the fork, and PRed into your dashing branch.  Let me know if you have any questions or prefer a different workflow.

pjreed, 2020-11-24:
Fixed by #604.

closed 2020-11-24 by pjreed