swri-robotics/marti_common #616: swri_transform_util: Transform manager needs transform listener in ROS2 foxy

jonselling, 2021-01-25:
The Transform Manager does not include transforms from the tf tree since it only uses a `tf2_ros::Buffer` and will also need a `tf2_ros::TransformListener` to receive transforms over the ROS tf topics.

I was able to fork the repo and create a workaround [here](https://github.com/jonselling/marti_common/tree/add-transform-listener).

danthony06, 2021-04-05:
Sorry for the slow response, I can start taking a look at this.

matt-attack, 2021-04-06:
How does mapviz work in ros2 without this?

pjreed, 2021-04-06:
The intended way to use this is to create a single TransformListener and Buffer, then pass the Buffer into your TransformManager.  Doing it this way is preferred so that a single process doesn't have multiple subscribers on a single topic and doesn't maintain multiple redundant buffers.

Here's how mapviz does it: https://github.com/swri-robotics/mapviz/blob/2477b3450cd6a698fea25ef9a9598969d3eeb424/mapviz/src/mapviz.cpp#L287

matt-attack, 2021-04-06:
Should we make a function to get and initialize (if necessary) a singleton transform manager then?

pjreed, 2021-04-06:
To a degree, I think the added bit of complication here is a good thing; the user should have to think about how they're managing their listener and buffer, because if they have anything using it other than a single TransformManager, carelessly creating multiple listeners and buffers will cause performance issues.  The buffer itself also needs to know [what type of clock you're using](https://docs.ros2.org/foxy/api/tf2_ros/classtf2__ros_1_1Buffer.html#a9ae70eade08c8ea7f6cc1b3ad3ec0d4f) and whether it has [a thread dedicated to it](https://docs.ros2.org/foxy/api/tf2/classtf2_1_1BufferCore.html#a1a6432ac9afc92689388f8c8f9e5dc23).  I'm not sure if there's a good way to hide all of that and still make it usable for anything but the most trivial of use cases.

jonselling, 2021-04-06:
I think understanding that the tf2 buffer will be filled external from the transform manager is fine as long as it is conveyed. It was a little confusing to use at first since the ROS 1 version starts the transform listener for you. Adding prints might be good enough -  [I think these lines](https://github.com/swri-robotics/marti_common/blob/56b32be63be3979244d406ec0cb8668c548da223/swri_transform_util/src/transform_manager.cpp#L170-L191). Maybe include some text that the issue could be the buffer is not being filled? It would be ideal if you could check the buffer for how many transforms it has directly, but that doesn't seem to be in the API.

matt-attack, 2021-04-06:
I think maintaining the same behavior by default as ros1 makes sense where it automatically generates the buffer and listener (the two are combined in tf1) unless one is specified.

danthony06, 2022-04-29:
@agyoungs did your recent PR incidentally fix this as well?

danthony06, 2022-05-25:
@jonselling I've made this PR which I think adds some of the warnings you're looking for. I also added an optional parameter so we can construct the transform manager with a buffer.

jonselling, 2022-05-25:
For reference, you are talking about #666

agyoungs, 2022-05-25:
> 

Just noticed this. No that PR would not have fixed this.

jonselling, 2022-05-25:
@danthony06 I think the changes are good to include but do not solve the issue described here. I believe the issue is that previously in ROS 1 the transform listener was created in this class. Now, with only the buffer, there is no way for someone to guarantee it gets filled like it did previously. 

I think resolution can be that just amending [these prints](https://github.com/swri-robotics/marti_common/blob/56b32be63be3979244d406ec0cb8668c548da223/swri_transform_util/src/transform_manager.cpp#L170-L191) that I link to earlier. I suggested this as a solution since there does not seem to be a public API for a buffer to get if it is empty or not.

danthony06, 2022-05-25:
Closed via: https://github.com/swri-robotics/marti_common/pull/666

closed 2022-05-25 by danthony06