swri-robotics/marti_common #620: [Foxy] swri_transform_util: UTMTransformer and Wgs84Transformer do not initialize correctly in transform_manager

jonselling, 2021-05-25:
When initializing the transform manager, the Wgs84Transformer  and UTMTransformer seem to fail when initializing:

```
[ERROR] [1621902479.930763076] [swri_transform_util::Transformer]: UtmTransformer::Initialize: No LocalXyWgs84Util has been set!
[ERROR] [1621902479.930777221] [swri_transform_util::Transformer]: Wgs84Transformer::Initialize: local_yx_util was unset!
```
I believe it is because each of them check that the `local_xy_util_` pointer is not null and that check fails. [Here is the UTM transformer check](https://github.com/swri-robotics/marti_common/blob/dashing-devel/swri_transform_util/src/utm_transformer.cpp#L140).

The `Initialize()` function is called from the Transformer parent class's Initialize function [here](https://github.com/swri-robotics/marti_common/blob/dashing-devel/swri_transform_util/src/transformer.cpp#L45).

The `local_xy_util_` pointer isn't set until *after* that function is called. 

As a side note, those transformers do accept a `local_xy_util` pointer in their constructor, but [they are set to null](https://github.com/swri-robotics/marti_common/blob/dashing-devel/swri_transform_util/src/transform_manager.cpp#L47-L48)

danthony06, 2021-05-25:
It kind of looks like the same issue that was raised here:

https://github.com/swri-robotics/marti_common/issues/619

jonselling, 2021-05-26:
You are correct from looking at the issue, but I feel like the resolution could easily be to change the order of the code in the Transformer's `Initialize()` function. There is really no reason to print an error for something that could be / is initialized shortly afterward.

danthony06, 2022-04-29:
@agyoungs have you ran into this?

jonselling, 2022-05-25:
I believe this will get fixed once #666 gets merged (due to re-ordering of the `Transformer::Initialize()` function)

danthony06, 2022-05-25:
Resolved via: https://github.com/swri-robotics/marti_common/pull/666

closed 2022-05-25 by danthony06