swri-robotics/marti_common #622: CMake Error: PROJ_LIBRARIES and swri_transform_util

joseandres15, 2021-06-24:
Hi everyone!

I am new using ROS and Linux in general, and I am trying to install this bag extractor for ROS Melodic (https://github.com/Robotics-Mechatronics-UMA/dataset_bag_extractor) in a Docker container.

I have installed all the dependencies that it needs. After installing more packages that it needs during the installation (GPS_common, swri_math_util, geographic _info,...) appears this error after trying `catkin_make`:

> CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
PROJ_LIBRARIES
    linked by target "swri_transform_util" in directory /opt/ros_ws/src/marti_common/swri_transform_util
libpcap_LIBRARIES
    linked by target "novatel_gps_driver" in directory /opt/ros_ws/src/novatel_gps_driver_modified/novatel_gps_driver

> -- Configuring incomplete, errors occurred!
See also "/opt/ros_ws/build/CMakeFiles/CMakeOutput.log".
See also "/opt/ros_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed

What should I do now? I have been looking in Google but I couldn't find something that could fix it.

pjreed, 2021-06-24:
Out of curiosity, how have you been installing packages?  It looks like you're still missing at least libproj and libpcap.  [rosdep](http://wiki.ros.org/rosdep) should take care of everything.  If you're not familiar with it, usage basically looks like

```bash
. /opt/ros/melodic/setup.bash
rosdep update
cd /opt/ros_ws
# To check which ROS dependencies you're missing in the current workspace:
rosdep check . --from-paths -i
# To install those dependencies:
rosdep install . -y --from-paths -i
```

joseandres15, 2021-06-24:
> Out of curiosity, how have you been installing packages? It looks like you're still missing at least libproj and libpcap. [rosdep](http://wiki.ros.org/rosdep) should take care of everything. If you're not familiar with it, usage basically looks like
> 
> ```shell
> . /opt/ros/melodic/setup.bash
> rosdep update
> cd /opt/ros_ws
> # To check which ROS dependencies you're missing in the current workspace:
> rosdep check . --from-paths -i
> # To install those dependencies:
> rosdep install . -y --from-paths -i
> ```

I have just done what you said and this is what I found:

After `rosdep check . --from-paths -i`:
> System dependencies have not been satisfied:
apt	libpcap0.8-dev
apt	libfmt-dev
apt	python-pyproj
apt	libproj-dev
apt	python3-opencv
apt	ros-melodic-camera-calibration-parsers
apt	python3-numpy
apt	python-psutil
ERROR[bag_extractor]: Cannot locate rosdep definition for [OpenCV]
	rosdep key : OpenCV
	OS name    : ubuntu
	OS version : bionic
	Data:
<no data>

After `rosdep install . -y --from-paths -i`
> ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
bag_extractor: Cannot locate rosdep definition for [OpenCV]

Also, following the tutorial for "marti_common", after doing `rosdep install --from-paths . --ignore-src`, I find this error:

> ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
bag_extractor: Cannot locate rosdep definition for [OpenCV]

The same as `rosdep install . -y --from-paths -i`

pjreed, 2021-06-24:
From that, I'm guessing that the `bag_extractor` package has a rosdep key named "OpenCV", but that is not a valid rosdep, so it's failing.  That package should probably depend on `cv_bridge`, or maybe `libopencv-dev` if it's using OpenCV but not any of the ROS bindings for it.

In the meantime, you can tell rosdep to skip that key by running:
```
rosdep install -y --from-paths -i --skip-keys=OpenCV
```

joseandres15, 2021-06-25:
> From that, I'm guessing that the `bag_extractor` package has a rosdep key named "OpenCV", but that is not a valid rosdep, so it's failing. That package should probably depend on `cv_bridge`, or [...]

I have tried `rosdep install -y --from-paths -i --skip-keys=OpenCV` and this appears:

>Usage: rosdep [options] <command> <args>
>
>Commands:
>
>rosdep check <stacks-and-packages>...
>  check if the dependencies of package(s) have been met.
>
>rosdep install <stacks-and-packages>...
>  download and install the dependencies of a given package or packages.
>
>rosdep db
>  generate the dependency database and print it to the console.
>
>rosdep init
>  initialize rosdep sources in /etc/ros/rosdep.  May require sudo.
>
>rosdep keys <stacks-and-packages>...
>  list the rosdep keys that the packages depend on.
>
>rosdep resolve <rosdeps>
>  resolve <rosdeps> to system dependencies
>
>rosdep update
>  update the local rosdep database based on the rosdep sources.
>
>rosdep what-needs <rosdeps>...
>  print a list of packages that declare a rosdep on (at least
>  one of) <rosdeps>
>
>rosdep where-defined <rosdeps>...
>  print a list of yaml files that declare a rosdep on (at least
>  one of) <rosdeps>
>
>rosdep fix-permissions
>  Recursively change the permissions of the user's ros home directory.
>  May require sudo.  Can be useful to fix permissions after calling
>  "rosdep update" with sudo accidentally.
>
>rosdep: error: no packages or stacks specified

So, I tried `rosdep install . -y --from-paths -i --skip-keys=OpenCV`, and this is what I found:

> ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
bag_extractor: Cannot locate rosdep definition for [Boost]

The same as before, but now it refers to Boost instead of OpenCV

Thanks so much for your help, btw

pjreed, 2021-06-25:
Yep, it looks like that's the same as before; `bag_extractor` says it depends on `Boost`, but that is not a valid rosdep key.  Try `rosdep install . -y --from-paths -i --skip-keys=OpenCV --skip-keys=Boost`.  I'm guessing the authors of `bag_extractor` are not familiar with rosdep, so you might have the repeat this process multiple times...

For reference, you can see a list of all known rosdep keys with `rosdep db`, which will print a (very long) list of keys and which packages they map to on your operating system.  If you're trying to find a particular package, it's useful to do something like `rosdep db | sort | grep -i boost` -- in which case we can see that `boost` maps to the `libboost-all-dev` package, which is probably what the authors of `bag_extractor` intended to depend on.

closed 2023-09-28 by danthony06