swri-robotics/marti_common #624: Warning: This header is deprecated

spam-12345, 2021-07-27:
Hi everyone,

when I build the package swri_image_util, I always get the warnings 

.... marti_common/swri_image_util/src/nodes/image_pub_node.cpp:35:
/opt/ros/foxy/include/image_transport/image_transport.h:41:89: note: #pragma message: Warning: This header is deprecated. Use 'image_transport.hpp' instead
   41 | #pragma message ("Warning: This header is deprecated. Use 'image_transport.hpp' instead")
....
....

for every included node.

Is there a way to restrain these warnings or can we fix it?

pjreed, 2021-07-27:
The annoying thing about this is that although this was deprecated in Foxy, the replacement, `image_transport.hpp`, was also not added until Foxy.  This means that "fixing" it by changing which header is included will break compatibility with pre-Foxy releases.  

For a while we had avoided changing this in order to support older distributions... but since Dashing reached EOL in May 2021, that means Foxy is now the oldest supported ROS2 distribution, so it's probably safe to change that include now.

danthony06, 2022-03-02:
This should be fixed now.

closed 2022-03-02 by danthony06