swri-robotics/marti_common #646: swri_transform_util 3.4.0 failing to build on ROS 2 build farm

nuclearsandwich, 2022-01-17:
* Package: `swri_transform_util`
* Version: 3.4.0-1
* ROS distro: Rolling
* Platforms:
  * [Ubuntu AMD64](https://build.ros2.org/job/Rbin_uF64__swri_transform_util__ubuntu_focal_amd64__binary/)
  * [Ubuntu ARM64](https://build.ros2.org/job/Rbin_ufv8_uFv8__swri_transform_util__ubuntu_focal_arm64__binary/)
  * [RHEL AMD64](https://build.ros2.org/job/Rbin_rhel_el864__swri_transform_util__rhel_8_x86_64__binary/)

This package is not currently building on any Rolling platforms. Below is an excerpt from the most recent [Ubuntu AMD64 build](https://build.ros2.org/job/Rbin_uF64__swri_transform_util__ubuntu_focal_amd64__binary/77/):

```
[100%] Linking CXX executable lat_lon_tf_echo
/usr/bin/cmake -E cmake_link_script CMakeFiles/lat_lon_tf_echo.dir/link.txt --verbose=1
/usr/lib/ccache/c++  -g -O2 -fdebug-prefix-map=/tmp/binarydeb/ros-rolling-swri-transform-util-3.4.0=. -fstack-protector-strong -Wformat -Werror=format-security -DNDEBUG -Wdate-time -D_FORTIFY_SOURCE=2   CMakeFiles/lat_lon_tf_echo.dir/src/nodes/lat_lon_tf_echo.cpp.o  -o lat_lon_tf_echo  -Wl,-rpath,/tmp/binarydeb/ros-rolling-swri-transform-util-3.4.0/obj-x86_64-linux-gnu:/opt/ros/rolling/lib:/opt/ros/rolling/lib/x86_64-linux-gnu: libswri_transform_util.so /usr/lib/x86_64-linux-gnu/libboost_filesystem.so.1.71.0 /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.71.0 /usr/lib/x86_64-linux-gnu/libboost_atomic.so.1.71.0 /usr/lib/x86_64-linux-gnu/libopencv_highgui.so.4.2.0 /usr/lib/x86_64-linux-gnu/libopencv_videoio.so.4.2.0 /usr/lib/x86_64-linux-gnu/libopencv_imgcodecs.so.4.2.0 /usr/lib/x86_64-linux-gnu/libopencv_video.so.4.2.0 /usr/lib/x86_64-linux-gnu/libopencv_calib3d.so.4.2.0 /usr/lib/x86_64-linux-gnu/libopencv_features2d.so.4.2.0 /usr/lib/x86_64-linux-gnu/libopencv_imgproc.so.4.2.0 /usr/lib/x86_64-linux-gnu/libopencv_flann.so.4.2.0 /usr/lib/x86_64-linux-gnu/libopencv_core.so.4.2.0 -lproj /usr/lib/x86_64-linux-gnu/libyaml-cpp.so.0.6.2 /opt/ros/rolling/lib/libdiagnostic_msgs__rosidl_typesupport_introspection_c.so /opt/ros/rolling/lib/libdiagnostic_msgs__rosidl_generator_c.so /opt/ros/rolling/lib/libdiagnostic_msgs__rosidl_typesupport_c.so /opt/ros/rolling/lib/libdiagnostic_msgs__rosidl_typesupport_introspection_cpp.so /opt/ros/rolling/lib/libdiagnostic_msgs__rosidl_typesupport_cpp.so /opt/ros/rolling/lib/libgeographic_msgs__rosidl_typesupport_introspection_c.so /opt/ros/rolling/lib/libgeographic_msgs__rosidl_generator_c.so /opt/ros/rolling/lib/libgeographic_msgs__rosidl_typesupport_c.so /opt/ros/rolling/lib/libgeographic_msgs__rosidl_typesupport_introspection_cpp.so /opt/ros/rolling/lib/libgeographic_msgs__rosidl_typesupport_cpp.so /opt/ros/rolling/lib/libgps_msgs__rosidl_typesupport_introspection_c.so /opt/ros/rolling/lib/libgps_msgs__rosidl_generator_c.so /opt/ros/rolling/lib/libgps_msgs__rosidl_typesupport_c.so /opt/ros/rolling/lib/libgps_msgs__rosidl_typesupport_introspection_cpp.so /opt/ros/rolling/lib/libgps_msgs__rosidl_typesupport_cpp.so /opt/ros/rolling/lib/libsensor_msgs__rosidl_typesupport_introspection_c.so /opt/ros/rolling/lib/libsensor_msgs__rosidl_generator_c.so /opt/ros/rolling/lib/libsensor_msgs__rosidl_typesupport_c.so /opt/ros/rolling/lib/libsensor_msgs__rosidl_typesupport_introspection_cpp.so /opt/ros/rolling/lib/libsensor_msgs__rosidl_typesupport_cpp.so /opt/ros/rolling/lib/libswri_roscpp__rosidl_typesupport_introspection_c.so /opt/ros/rolling/lib/libswri_roscpp__rosidl_generator_c.so /opt/ros/rolling/lib/libswri_roscpp__rosidl_typesupport_c.so /opt/ros/rolling/lib/libswri_roscpp__rosidl_typesupport_introspection_cpp.so /opt/ros/rolling/lib/libswri_roscpp__rosidl_typesupport_cpp.so /opt/ros/rolling/lib/libmarti_common_msgs__rosidl_typesupport_introspection_c.so /opt/ros/rolling/lib/libmarti_common_msgs__rosidl_generator_c.so /opt/ros/rolling/lib/libmarti_common_msgs__rosidl_typesupport_c.so /opt/ros/rolling/lib/libmarti_common_msgs__rosidl_typesupport_introspection_cpp.so /opt/ros/rolling/lib/libmarti_common_msgs__rosidl_typesupport_cpp.so /opt/ros/rolling/lib/liborocos-kdl.so.1.4.0 /opt/ros/rolling/lib/libstatic_transform_broadcaster_node.so /opt/ros/rolling/lib/libtf2_ros.so /opt/ros/rolling/lib/libtf2.so /opt/ros/rolling/lib/libmessage_filters.so /opt/ros/rolling/lib/librclcpp_action.so /opt/ros/rolling/lib/librcl_action.so /opt/ros/rolling/lib/libtf2_msgs__rosidl_typesupport_introspection_c.so /opt/ros/rolling/lib/libtf2_msgs__rosidl_generator_c.so /opt/ros/rolling/lib/libtf2_msgs__rosidl_typesupport_c.so /opt/ros/rolling/lib/libtf2_msgs__rosidl_typesupport_introspection_cpp.so /opt/ros/rolling/lib/libtf2_msgs__rosidl_typesupport_cpp.so /opt/ros/rolling/lib/libgeometry_msgs__rosidl_typesupport_introspection_c.so /opt/ros/rolling/lib/libgeometry_msgs__rosidl_generator_c.so /opt/ros/rolling/lib/libgeometry_msgs__rosidl_typesupport_c.so /opt/ros/rolling/lib/libgeometry_msgs__rosidl_typesupport_introspection_cpp.so /opt/ros/rolling/lib/libgeometry_msgs__rosidl_typesupport_cpp.so /opt/ros/rolling/lib/libaction_msgs__rosidl_typesupport_introspection_c.so /opt/ros/rolling/lib/libaction_msgs__rosidl_generator_c.so /opt/ros/rolling/lib/libaction_msgs__rosidl_typesupport_c.so /opt/ros/rolling/lib/libaction_msgs__rosidl_typesupport_introspection_cpp.so /opt/ros/rolling/lib/libaction_msgs__rosidl_typesupport_cpp.so /opt/ros/rolling/lib/libunique_identifier_msgs__rosidl_typesupport_introspection_c.so /opt/ros/rolling/lib/libunique_identifier_msgs__rosidl_generator_c.so /opt/ros/rolling/lib/libunique_identifier_msgs__rosidl_typesupport_c.so /opt/ros/rolling/lib/libunique_identifier_msgs__rosidl_typesupport_introspection_cpp.so /opt/ros/rolling/lib/libunique_identifier_msgs__rosidl_typesupport_cpp.so /opt/ros/rolling/lib/libcomponent_manager.so /opt/ros/rolling/lib/librclcpp.so /opt/ros/rolling/lib/libclass_loader.so /opt/ros/rolling/lib/x86_64-linux-gnu/libconsole_bridge.so.1.0 /opt/ros/rolling/lib/libcomposition_interfaces__rosidl_typesupport_introspection_c.so /opt/ros/rolling/lib/libcomposition_interfaces__rosidl_generator_c.so /opt/ros/rolling/lib/libcomposition_interfaces__rosidl_typesupport_c.so /opt/ros/rolling/lib/libcomposition_interfaces__rosidl_typesupport_introspection_cpp.so /opt/ros/rolling/lib/libcomposition_interfaces__rosidl_typesupport_cpp.so /opt/ros/rolling/lib/librclcpp.so -lpthread /opt/ros/rolling/lib/liblibstatistics_collector.so /opt/ros/rolling/lib/liblibstatistics_collector_test_msgs__rosidl_typesupport_introspection_c.so /opt/ros/rolling/lib/liblibstatistics_collector_test_msgs__rosidl_generator_c.so /opt/ros/rolling/lib/liblibstatistics_collector_test_msgs__rosidl_typesupport_c.so /opt/ros/rolling/lib/liblibstatistics_collector_test_msgs__rosidl_typesupport_introspection_cpp.so /opt/ros/rolling/lib/liblibstatistics_collector_test_msgs__rosidl_typesupport_cpp.so /opt/ros/rolling/lib/libstd_msgs__rosidl_typesupport_introspection_c.so /opt/ros/rolling/lib/libstd_msgs__rosidl_generator_c.so /opt/ros/rolling/lib/libstd_msgs__rosidl_typesupport_c.so /opt/ros/rolling/lib/libstd_msgs__rosidl_typesupport_introspection_cpp.so /opt/ros/rolling/lib/libstd_msgs__rosidl_typesupport_cpp.so /opt/ros/rolling/lib/librcl.so /opt/ros/rolling/lib/librcl_interfaces__rosidl_typesupport_introspection_c.so /opt/ros/rolling/lib/librcl_interfaces__rosidl_generator_c.so /opt/ros/rolling/lib/librcl_interfaces__rosidl_typesupport_c.so /opt/ros/rolling/lib/librcl_interfaces__rosidl_typesupport_introspection_cpp.so /opt/ros/rolling/lib/librcl_interfaces__rosidl_typesupport_cpp.so /opt/ros/rolling/lib/librmw_implementation.so /opt/ros/rolling/lib/libament_index_cpp.so /opt/ros/rolling/lib/librcl_logging_spdlog.so /opt/ros/rolling/lib/librcl_logging_interface.so /opt/ros/rolling/lib/librcl_yaml_param_parser.so /opt/ros/rolling/lib/librmw.so /opt/ros/rolling/lib/libyaml.so /opt/ros/rolling/lib/librosgraph_msgs__rosidl_typesupport_introspection_c.so /opt/ros/rolling/lib/librosgraph_msgs__rosidl_generator_c.so /opt/ros/rolling/lib/librosgraph_msgs__rosidl_typesupport_c.so /opt/ros/rolling/lib/librosgraph_msgs__rosidl_typesupport_introspection_cpp.so /opt/ros/rolling/lib/librosgraph_msgs__rosidl_typesupport_cpp.so /opt/ros/rolling/lib/libstatistics_msgs__rosidl_typesupport_introspection_c.so /opt/ros/rolling/lib/libstatistics_msgs__rosidl_generator_c.so /opt/ros/rolling/lib/libstatistics_msgs__rosidl_typesupport_c.so /opt/ros/rolling/lib/libstatistics_msgs__rosidl_typesupport_introspection_cpp.so /opt/ros/rolling/lib/libstatistics_msgs__rosidl_typesupport_cpp.so /opt/ros/rolling/lib/libb [... cut ...]
make[4]: *** [CMakeFiles/lat_lon_tf_echo.dir/build.make:216: lat_lon_tf_echo] Error 1
make[4]: Leaving directory '/tmp/binarydeb/ros-rolling-swri-transform-util-3.4.0/obj-x86_64-linux-gnu'
make[3]: *** [CMakeFiles/Makefile2:250: CMakeFiles/lat_lon_tf_echo.dir/all] Error 2
make[3]: Leaving directory '/tmp/binarydeb/ros-rolling-swri-transform-util-3.4.0/obj-x86_64-linux-gnu'
make[2]: Leaving directory '/tmp/binarydeb/ros-rolling-swri-transform-util-3.4.0/obj-x86_64-linux-gnu'
make[2]: *** [Makefile:144: all] Error 2
```

danthony06, 2022-01-17:
We'll take a look.

danthony06, 2022-02-01:
Looking now.

danthony06, 2022-02-04:
@nuclearsandwich I think we're hitting this issue, which still doesn't appear to have a great resolution:
https://github.com/ros2/geometry2/issues/242

danthony06, 2022-02-04:
Looks like it's the same issue as here: https://github.com/rst-tu-dortmund/teb_local_planner/issues/334, which will be fixed when https://github.com/ros2/geometry2/pull/485 gets merged.

danthony06, 2022-04-29:
Closing this now, we're going to push new releases, and the 3.4.1 releases appear to have built for Foxy and Galactic, while Rolling has a different issue.