swri-robotics/marti_common #660: bad_weak_ptr with swri_image_util (ros2-devel)

agyoungs, 2022-04-28:
`shared_from_this()` cannot be used in constructors since it will cause a `bad_weak_ptr` runtime exception.

I fixed this in a less than ideal way (checking at each use) in #659, however there was one node I didn't touch since there wasn't a way to solve this with the same method.

[replace_colors_node.cpp](https://github.com/swri-robotics/marti_common/blob/ros2-devel/swri_image_util/src/nodes/replace_colors_node.cpp) uses image transport and passes the `Node::SharedPtr` in the constructor. I think this could be a normal `rclcpp::Subscription` instead of an `image_transport` subscription, but before I created a PR I wanted to see what others thought.

    // Set up the ROS interface
    image_transport::ImageTransport it(shared_from_this());
    image_pub_ = it.advertise("modified_image", 1);
    image_sub_ = it.subscribe(
      "image",
      1,
      &ReplaceColorsNodelet::imageCallback,
      this);

matt-attack, 2022-04-28:
There's not really a great solution here since ROS removed the initialize function from the "nodelet base class" in ROS2. 

I think the issue in this particular case is that image transport is taking the node as a shared pointer at all. Taking a shared pointer implies that it is taking ownership of the node, which is very much not the case (at least in this application) unless you love memory leaks.

matt-attack, 2022-04-29:
Though, I'm now just realizing you can probably actually just store the logger in the initializer for swri_transform_manager and avoid needing to change the lookup function

matt-attack, 2022-04-29:
The other possible solution, which may or may not be possible (and if so probably annoying) is not inheriting from Node in components at all. From a brief search, it seems possible but at the least you'd have to set up your own executor.

https://ubuntu.com/blog/components-vs-plugins-in-ros-2