swri-robotics/marti_common #665: Package build failures in ROS 2 Humble and Rolling

nuclearsandwich, 2022-05-24:
Some packages from this repository are failing to build in ROS 2 Rolling and ROS 2 Humble.
Other packages from this repository are unavailable due to the build issues with their sibling packages, which they depend on.

These packages are failing directly:
* swri_console_util
* swri_geometry_util
* swri_roscpp
* swri_prefix_tools (RHEL 8 only, fix unreleased in https://github.com/swri-robotics/marti_common/pull/652)

Packages failing to build are going to be blacklisted on the build farm which will disable the recurring failing builds. If a future release resolves the failures then the packages can be removed from the blacklist.

nuclearsandwich, 2022-05-24:
I've proposed blacklisting these packages pending an updated release in https://github.com/ros2/ros_buildfarm_config/pull/240

danthony06, 2022-10-10:
@nuclearsandwich We've updated for Humble. I'm having troubles blooming the packages, presumably because of the blacklist. Could I get them unblocked please?

cottsay, 2022-10-11:
> I'm having troubles blooming the packages, presumably because of the blacklist.

The `ros_buildfarm_config` blacklist isn't part of the Blooming process at all. We need to get the updated packages bloomed and merged into `ros/rosdistro` before we can unblock the builds.

danthony06, 2023-09-21:
@cottsay our packages still aren't getting released. It looks like they're still getting blacklisted here: https://github.com/ros2/ros_buildfarm_config/edit/ros2/humble/release-build.yaml for Humble, even though they've been merged into rosdistro. Could we get them going again?

closed 2023-09-28 by danthony06