swri-robotics/marti_common #673: Not all packages released for ROS Humble

ksuszka, 2022-07-08:
As on 2022-07-08 not all core packages of swri are available in apt on Ubuntu 22.04, although info on https://index.ros.org/r/marti_common/#humble states they were released.
To reproduce run: 
``` shell
$ docker run -it ros:humble bash -c "apt update && apt search ros-humble-swri"
...
ros-humble-swri-dbw-interface/jammy 3.4.0-3jammy.20220519.235534 amd64
  This package provides documentation on common interface conventions for drive-by-wire systems.

ros-humble-swri-math-util/jammy 3.4.0-3jammy.20220607.133833 amd64
  swri_math_util

ros-humble-swri-math-util-dbgsym/jammy 3.4.0-3jammy.20220607.133833 amd64
  debug symbols for ros-humble-swri-math-util

ros-humble-swri-opencv-util/jammy 3.4.0-3jammy.20220607.141100 amd64
  swri_opencv_util

ros-humble-swri-prefix-tools/jammy 3.4.0-3jammy.20220519.235538 amd64
  Contains scripts that are useful as prefix commands for nodes started by roslaunch.

ros-humble-swri-serial-util/jammy 3.4.0-3jammy.20220519.235545 amd64
  swri_serial_util

ros-humble-swri-system-util/jammy 3.4.0-3jammy.20220607.133918 amd64
  swri_system_util
```
Some crucial packages like `ros-humble-swri-roscpp` are missing.

ksuszka, 2022-07-08:
I've just noticed it is probably due to https://github.com/swri-robotics/marti_common/issues/665

danthony06, 2022-10-10:
@ksuszka v3.4.2 built on the ```ros2-devel``` branch is now working with Humble. I'm working on releasing it to the ROS2 buildfarm for packaging.

closed 2022-10-10 by danthony06