swri-robotics/marti_common #695: UTM to Lat-Long function returning incorrect Values

dstoianovici, 2022-11-07:
When using the `UtmToTfTransform::Transform()` frunction from the latest release of `swri_transform_util` on the `ros2-devel` branch, the resulting frame transforms from UTM to LatLong appear to be broken.

 This was tested by adding print statements in `utm_transformer.cpp` at line 182 to print the input UTM coordinates, and comparing output LatLong coordinates to calculated results. Input UTM coordinates and resulting LatLong coordinates are not equal.

It appears that the transformation performed by `UtmUtil::UtmData::ToLatLon()` is calculating incorrectly.

 In order to fix this in use, the `ros2-devel` branch was reverted back to commit `5fc6ed0ea39faa70f49cf0c0dc94bdc895d8b94e`, before support for ROS2 Humble was added.

danthony06, 2022-11-29:
Can you provide an example of the inputs and expected outputs? We have a few test cases in `swri_transform_util`, and they're still passing.

danthony06, 2023-09-29:
I don't think I can do much with this without a specific test case. We've been running the `ros2-devel` branch on quite a few vehicles and haven't seen problems with the UTM to WGS84 conversion, so we can't reproduce the problem.

closed 2023-09-29 by danthony06