swri-robotics/marti_common #703: [Humble]: swri_roscpp does not export dependencies correctly

jonselling, 2023-02-24:
# Description

When building against `swri_roscpp` in ROS2 humble, there is an error that the `diagnostic_updater` is not exported correctly as a dependency (at least for finding the header file). I get this error unless I add the dependency to my own package:

```
install/swri_roscpp/include/swri_roscpp/subscriber.h:34:10: fatal error: diagnostic_updater/diagnostic_status_wrapper.hpp: No such file or directory
   34 | #include <diagnostic_updater/diagnostic_status_wrapper.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

# Environment

Branch: ros2-devel
Distro: humble

# Actual Results

I get a build error against a dependency my package does not use

# Expected Results

That dependency should get passed as an export from swri_roscpp to my package.

# Additional Info

I checked the CMakeLists.txt and it looks like the dependency _should_ get exported, so I started to mess around with the file a little.

I was able to compile with removing any of the calls that use the `${PROJECT_NAME}` interface and then removing the `BUILD_TESTING` block. 

I am unsure if it is because the ${PROJECT_NAME} export data gets written weirdly with the INTERFACE target and then the `rosidl_generate_interfaces` stuff might interfere in the testing part of the CMakeLists.txt.

danthony06, 2023-02-24:
I'll take a look at what's going on.

danthony06, 2023-03-13:
@jonselling when you said you removed any of the calls that use the `${PROJECT_NAME}` interface, what did you mean by that?

jonselling, 2023-03-13:
I commented out these lines in the CMakeLists.txt
```
add_library(${PROJECT_NAME} INTERFACE)

target_include_directories(${PROJECT_NAME} INTERFACE
  "$<BUILD_INTERFACE:${CMAKE_CURRENT_SOURCE_DIR}/include>"
  "$<INSTALL_INTERFACE:include>"
)

install(TARGETS ${PROJECT_NAME}
  EXPORT export_${PROJECT_NAME}
  ARCHIVE DESTINATION lib
  LIBRARY DESTINATION lib
  RUNTIME DESTINATION lib
  INCLUDES DESTINATION include
)

...

ament_export_targets(export_${PROJECT_NAME} HAS_LIBRARY_TARGET)
```

And then I also commented out everything in the `if (BUILD_TESTING AND ...)` statement

closed 2023-04-04 by danthony06