swri-robotics/marti_common #759: [Noetic] swri_yaml_util "Unknown arguments specified" build failure

sloretz, 2025-02-26:
The swri_yaml_util package has started failing on the ROS Noetic buildfarm, and that causes a few regressions: https://repositories.ros.org/status_page/ros_noetic_default.html?q=REGRESSION

https://build.ros.org/view/Nbin_uF64/job/Nbin_uF64__swri_yaml_util__ubuntu_focal_amd64__binary/18/console

```
20:53:40 CMake Error at CMakeLists.txt:45 (if):
20:53:40   if given arguments:
20:53:40 
20:53:40     "STREQUAL" "kinetic"
20:53:40 
20:53:40   Unknown arguments specified
20:53:40 
20:53:40 
20:53:40 -- Configuring incomplete, errors occurred!
```

The code in question: https://github.com/swri-robotics/marti_common/blob/523c8aa6b8b40c2c118b0ead8333ecdfd02e1c95/swri_yaml_util/CMakeLists.txt#L45-L49

[Noetic requires C++ 14](https://www.ros.org/reps/rep-0003.html#noetic-ninjemys-may-2020-may-2025). I think this could be fixed by making that block unconditional as is, or by unconditionally using:

```
target_compile_features(${PROJECT_NAME} PUBLIC cxx_std_14)
```

danthony06, 2025-02-26:
Sorry about that, I'll take a look tonight and get it patched.

danthony06, 2025-02-27:
Just bloomed 2.15.4 that removed the conditional check on ROS_DISTRO and set the C++ standard to C++14 as you suggested. PR https://github.com/swri-robotics/marti_common/pull/760 should resolve this issue.

closed 2025-02-27 by danthony06